avaldebe/PMserial #2: Sleep mode

szerwi, 2018-09-19:
How to turn on sleep mode in this sensor?

avaldebe, 2018-09-20:
It should be on passive mode between reads, but I'm not sure it really works. Not all available datasheets show the configuration commands. If you want to be sure the sensor goes to sleep, you could ground the SET pin (3rd pin on the header).

szerwi, 2018-09-20:
I've tried to ground 3 pin, but it did not work. The fan in the sensor was still working.

avaldebe, 2018-09-20:
That is strange. Which sensor are you working with?

szerwi, 2018-09-20:
I'm trying to modify your library and send sleep/wake commands. According to datasheet (http://www.aqmd.gov/docs/default-source/aq-spec/resources-page/plantower-pms5003-manual_v2-3.pdf) it should look like this:
Sleep: slp[msgLen] = {0x42,0x4D,0xE4,0x00,0x00,0x01,0x72}
Wake: wak[msgLen] = {0x42,0x4D,0xE4,0x00,0x01,0x01,0x72};

Functions in PMserial.cpp:
void SerialPM::sleep() {
  trigRead(slp,msgLen);  // sleep
}

void SerialPM::wake() {
  trigRead(wak,msgLen);  // sleep
}

But the sensor is still working after pms.sleep();. Am I doing something wrong?

avaldebe, 2018-09-21:
It looms like it should work. Maybe the CRC (cleck code, last 2 bytes of the message) are wrong, and the sensor is rejecting the command. As I wrote before, I'm not sure that the PMSx003 sensor actually accepts the commands the library is sending.
Alas, I wont have time until next weekend to look into this issue in hardware.

szerwi, 2018-09-21:
@avaldebe it worked. Thank you for your help :)

closed 2018-09-21 by szerwi