avaldebe/PMserial #3: General docs/usage

avaldebe, 2019-04-17:
If you do not understand how to use the library, found an explanation confusing or you found a typo. Leave a message and will try to answer as soon as reasonable.

Thanks for helping to improve the library!

SFeli, 2020-07-26:
Hi - thx for the greate library - have moved over to your library because auf time.h - incompatibilities of the alternative.
HW: PMS7003 / Lolin D32 (ESP32)
SW: VSCode with PIO
### Note 1) ESP-Serial issue. The following code works and is flexible for alternative pins

```
HarwareSerial pmsSerial(2); 
SerialPM pms(PMSx003, pmsSerial); 
in setup()
    pmsSerial.begin(9600, SERIAL_8N1, 16, 17)
```    

### Note 2) Sleep
In Issue 2 [https://github.com/avaldebe/PMserial/issues/2](url) you mention an workaround by edditing the library with 
```
void SerialPM::sleep() {
trigRead(slp,msgLen); // sleep
}

void SerialPM::wake() {
trigRead(wak,msgLen); // sleep
}
```
**but** trigRead() has as fare as I can see no parameters. Perhaps you have to add:
```
void SerialPM::wake() {
  uart->write(wak, msgLen); // wakeup
  uart->flush();
  delay(max_wait_ms * 2);
  while (uart->available())
  {
    uart->read(); // empty the RX buffer
  }
}
```
Perhaps you can include this function in the standard library or add this option in issue 2.

Thanks for your work and congratulation for the efficient lines :-)

avaldebe, 2020-07-27:
Hi @SFeli 

Thanks fort taking an interest on my library.

> ### Note 1) ESP-Serial issue
> The following code works and is flexible for alternative pins
> 
> ```c++
> HarwareSerial pmsSerial(2); 
> SerialPM pms(PMSx003, pmsSerial); 
> [...]
> setup(){
>     [...]
>     pmsSerial.begin(9600, SERIAL_8N1, 16, 17)
> }
> ```

I think that this use case was addressed on #7. This library uses Serial1 on the ESP32 to connect a PM sensor on user defined pins. Probably the [explanation](https://github.com/avaldebe/PMserial#esp32-serial1) is not clear enough.

avaldebe, 2020-07-27:
> ### Note 2) Sleep
> In #2 you mention an workaround by editing the library with
> 
> ```c++
> void SerialPM::sleep() {
> trigRead(slp,msgLen); // sleep
> }
> 
> void SerialPM::wake() {
> trigRead(wak,msgLen); // sleep
> }
> ```
> 
> **but** trigRead() has as fare as I can see no parameters.

This [code snippet](https://github.com/avaldebe/PMserial/issues/2#issuecomment-423259172) is not mine, It was part of an attempt by a library user to implement sleep.
I never reviewed the whole code, but I guess that it included changes to `trigRead()`.

Do you need/want sleep/wake modes?
IMO, the best way to put the sensor to sleep is using the sensor SET pin.

MobeenAshraf, 2021-08-21:
Can we have some more documentation around how to setup and steps to use this code?
As a beginner, I think it uses platformio but coming from ArduinoIDE setup, Can we add more steps on what needs to be done to install this code on Controller?

avaldebe, 2021-08-23:
Hi @MobeenAshraf 

> Can we have some more documentation around how to setup and steps to use this code?

In addition to the [examples](https://github.com/avaldebe/PMserial/tree/master/examples)?

> As a beginner, I think it uses platformio but coming from ArduinoIDE setup, Can we add more steps on what needs to be done to install this code on Controller?

I have not used the Arduino IDE on years, back on the days before the Library Manager, so I do not know how to add libraries to the IDE. In any case, I think there are plenty of tutorials that teach how to do it, e.g.:

- https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use/library-manager
- https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use/how-to-install-a-library

I think I could add some links to the library README. Please let me know if you have a better idea.

> Also, for beginner and minimal setup can we add minimal circuit diagram links aswell?

I included links to the datasheets I could find for every supported sensors, but I guess this is not enough for a beginner.
I'll need to think about this, as I do not want to write a full tutorial for this sensor (there are quite a few already).
However, I would be glad to include documentation submitted via PR (hint hint).