avaldebe/PMserial #4: Data logger example

avaldebe, 2019-04-17:
You found this library useful and would like to give something back. Here is your chance.
I would appreciate more example projects. PRs are welcomed.

## Outline

- Take measurements at regular intervals.
- Sleep/low power mode in between measurements.
- Save measurements to  external FLASH or an SD CARD.
- Timestamp measurements with any of the following options
  - external RTC such as DS3231
  - internal RTC in STM32 or SAMD21  boards
  - NTP for boards with network access such as ESP8266 or ESP32 boards

gbalbinot, 2019-10-22:
Hi Alvaro,

We started using your library in our devices (5 so far) in a project to monitor Air Quality in Porto Alegre, Rio Grande do Sul, Brasil. It solved our issues with spur readings, it is working very well. Congratulations on your excellent work. 

We measure every 4 minutes, no sleep between measurements. 

Do you plan on supporting PMS5003T ?

Regards

gbalbinot, 2019-10-22:
also, if I download the files .zip and try to import in arduino studio, it does not work. I believe it is missing the library.properties file.

avaldebe, 2019-10-22:
Hi @gbalbinot 

> We started using your library in our devices (5 so far) in a project to monitor Air Quality in Porto Alegre, Rio Grande do Sul, Brasil. It solved our issues with spur readings, it is working very well. Congratulations on your excellent work.

Thanks!

> We measure every 4 minutes, no sleep between measurements. We use ESP32.

I also keep my sensors awake between measurements. My experiments with sleep/wake a `PMSA003` sensor, showed that the measurements were not reliable just after the sensor wakes up.

> Do you plan on supporting PMS5003T ?

I have none of those devices to test the library with. I should be able to add support, but it would be messy.

Recently added support for the `PMS5003T` on [my Python data acquisition application][pypms]
from from information I found on a `PMS5003T` datasheet.
The `PMS5003T` replaces the last 2 number concentrations (`n5p0` and `n10p0`)
with `temp*10` and `rel.hum*10`.
This was easy to implement on Python, but can not think of a clean way to implement it on C++.

[pypms]: https://github.com/avaldebe/PyPMS

Maybe you could try something like the following?

```arduino
  Serial.printf("PM1.0 %2d, PM2.5 %2d, PM10 %2d [ug/m3]\n",
    pms.pm01,pms.pm25,pms.pm10);
  Serial.printf("N0.3 %4d, N0.5 %3d, N1.0 %2d, N2.5 %2d [#/100cc]\n",
    pms.n0p3,pms.n0p5,pms.n1p0,pms.n2p5);
  Serial.printf("Temp. %5.1f [*C], Rel.Hum. %5.1f [%]\n",
    pms.n5p0/10.0,pms.n10p0/10.0);
```

Let me know if it works, or open an issue asking for `PMS5003T` support.
Alas, It might take me some time to implement it.

> Also, if I download the files .zip and try to import in arduino studio, it does not work. I believe it is missing the library.properties file.

I left the Ardhino IDE long time ago and have never used arduino studio.
Can you show me a `library.properties` file?

Cheers

gbalbinot, 2020-04-16:
Hi Alvaro,

Thank you for the PMS5003T update. Actually I'm not sure how to make it
"importable" in arduino studio, so I have no idea of library.properties.

I could not test your proposed solution as we don't have any PMS5003T units
yet.

Cheers,
Guilherme