avaldebe/PMserial #5: Need option to disable sending command to sensor

tms320, 2019-12-17:
Hello!
I use the same UART channel to read data from sensor (via RX pin of controller) and send debug messages (via TX pin). So there is no connection between controller's TX pin and sensor's RX pin.
In my case PMS5003 sensor works fine without sending any command to it.
Your library sends "passive mode" command in 'init' and 'trigRead' methods:
`uart->write(cfg, msgLen); // set passive mode`
so I see this data as unreadable text in my debug monitor.
Is it possible to add some option (in constructor, for example) to disable sending any data to UART?
Thanks )

tms320, 2019-12-22:
Closed. I decided to use SoftwareSerial.

avaldebe, 2019-12-23:
Sorry for the delayed answer, looks like you found a way around the issue.

> Your library sends "passive mode" command in 'init' and 'trigRead' methods:
uart->write(cfg, msgLen); // set passive mode
so I see this data as unreadable text in my debug monitor.
Is it possible to add some option (in constructor, for example) to disable sending any data to UART?

Alas, this messages are needed for reading the sensor data on demand (passive mode operation).
Otherwise the sensor will send a new set of measurements every ~800 ms (active mode operation). 

I wrote this library for passive mode operation. When I wrote the library there were already other libraries available that dealt with active model operations, so I did not addressed the active mode usage.
The PMS3003 does not support passive mode operation, so I built the library to work around this limitation. Therefore, implementing an `active_mode` option should be relatively easy.

Thanks for giving my library a try. I'll reopen the issue,
in case someone else wants an `active_mode` option (PRs are most welcomed)

tms320, 2019-12-23:
In PMS5003 datasheet that I have there is no information about passive/active mode and any commands to sensor. Thanks for your explanations )

avaldebe, 2019-12-27:
There are several versions of the datasheet. This one was translated from the original Chinese one, is the best reference I have found

http://www.aqmd.gov/docs/default-source/aq-spec/resources-page/plantower-pms5003-manual_v2-3.pdf?sfvrsn=2

tms320, 2019-12-27:
Thanks a lot )

closed 2025-11-07 by avaldebe