avaldebe/PMserial #6: doesn't work with 7003

yburkouski, 2019-12-25:
Doesn't work with 7003 under Arduino

avaldebe, 2019-12-27:
Will look into this, hopefully sometime this weekend.
In the meantime, can you give me some extra information?
Did you try any of the library examples?
Which Arduino board are you using?

pczekalski, 2020-01-28:
Hi, same here "Incomplete message header". The sensor is early PMS7003.
Just started testing with Softserial on ESP8266-12E (NodeMcu 1.0).
I do make this work as I wrote a library in C# and connected via USB to Serial converter to my PC so I'm absolutely sure it is OK. What I found is that sometimes it used to send "shortened" frame. I didn't analyze your library but what happens if you once discover invalid header? Do you just drop the frame and wait for the starting flag?

avaldebe, 2020-01-29:
> Will look into this, hopefully sometime this weekend.

Alas, I'm yet to find time to look into this issue.

> I didn't analyze your library but what happens if you once discover invalid header? Do you just drop the frame and wait for the starting flag?

Invalid messages are drop. The status returned should tell what was the problem.
The SoftwareSerial example shows how to interpret the return status

https://github.com/avaldebe/PMserial/blob/55559252e6318503e1113c20930f08bdc6d4e694/examples/SoftwareSerial/SoftwareSerial.ino#L35-L63

Now, I'm not sure what would be the best way to deal with thins kind of errors.
Maybe could add an optional retry argument to the read method.

strange-v, 2020-02-01:
I've just started to use this library and it does work with PMS7003 on ESP32 (hardware serial).
AFAIK, ESP8266 has some issues with software serial.

pczekalski, 2020-02-02:
Hard to say. I'm using another library but on soft serial as well, and it
works seamlessly. So that is specific combination of soft serial, this
library and ESP8266.

Matz88, 2020-04-15:
Same issue here with ESP8266, "Incomplete message header", you were mentioning the possibility to add a retry in the read, could you further explain? I can try

Matz88, 2020-04-15:
Further update: I have disabled one by one all the error triggered by the library to let the SW run till the end. I can see the raw numbers changing but the calibrated numbers are never !=0:

12:35:00.747 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:35:10.697 ->  424D 0014 0000 0002 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:35:10.766 -> Errors 1/4, waited 10 ms
12:35:10.800 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:35:20.736 ->  424D 0014 0001 0002 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:35:20.805 -> Errors 1/5, waited 10 ms
12:35:20.839 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:35:30.770 ->  424D 0014 0001 0002 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:35:30.838 -> Errors 1/6, waited 11 ms
12:35:30.873 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:35:40.782 ->  424D 0014 0001 0002 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:35:40.851 -> Errors 1/7, waited 10 ms
12:35:40.886 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:35:50.834 ->  424D 0014 0002 0003 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:35:50.902 -> Errors 1/8, waited 10 ms
12:35:50.937 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:00.853 ->  424D 0014 0024 00AC 0100 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:00.922 -> Errors 1/9, waited 10 ms
12:36:00.957 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:10.910 ->  424D 0014 0033 00CA 0100 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:10.978 -> Errors 1/10, waited 10 ms
12:36:11.012 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:20.935 ->  424D 0014 0035 00CD 0100 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:21.005 -> Errors 1/11, waited 10 ms
12:36:21.040 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:30.953 ->  424D 0014 0016 002B 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:31.023 -> Errors 1/12, waited 10 ms
12:36:31.057 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:41.014 ->  424D 0014 0007 000E 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:41.049 -> Errors 1/13, waited 10 ms
12:36:41.084 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]
12:36:51.025 ->  424D 0014 0006 000C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
12:36:51.095 -> Errors 1/14, waited 10 ms
12:36:51.130 -> PM1.0  0, PM2.5  0, PM10  0 [ug/m3]

avaldebe, 2020-04-15:
Looks like it goes deeper than this library

Here is the PMSA003 output of the debug sketch for 

```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
build_flags = -D PMS_RX=D7 -D PMS_TX=D6
```
```
PMSx003 on SoftwareSerial
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 1/1, waited 1004 ms
Sensor read timeout
 424D 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 2/2, waited 780 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 3/3, waited 783 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 4/4, waited 423 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 5/5, waited 422 ms
Incomplete message header
```

increasing the waiting time for a complete message from 1s to 5s deals with the timeout error,
but it fails to decode

```
PMSx003 on SoftwareSerial
 424D 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 1/1, waited 2742 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 2/2, waited 850 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 3/3, waited 712 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 4/4, waited 424 ms
Incomplete message header
 424D 001C 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
Errors 5/5, waited 422 ms
Incomplete message header
```

The header is incomplete on the 1st message, but the next ones look fine.

avaldebe, 2020-04-15:
Looks like the failed decoding with espressif8266@2.4.0 is related to plerup/espsoftwareserial#63
downgrading the platform to espressif8266@2.3.0 or upgrading underlying SoftwareSerial library to EspSoftwareSerial@>=6.7.1 seems to work

```ini
;  downgrade the platform
[env:d1_mini]
platform = espressif8266@2.3.0
board = d1_mini
build_flags = -D PMS_RX=D7 -D PMS_TX=D6
```

```ini
;  upgrade SoftwareSerial
[env:d1_mini]
platform = espressif8266
lib_deps = EspSoftwareSerial@>=6.7.1
board = d1_mini
build_flags = -D PMS_RX=D7 -D PMS_TX=D6
```

Matz88, 2020-04-15:
Thanks, worked after downloading the latest SWserial and killing the one under AppData\Local\ !

PR #9 (2020-04-15): SoftwareSerial on ESP boards
Fix broken SoftwareSerial on ESP8266, close #6 
~~Add SoftwareSerial support for the ESP32, see #7~~
closed 2020-04-15 by avaldebe