avaldebe/PMserial #8: Support for Arduino MKRWifi 1010

anass-naouss, 2020-04-15:
Hi avaldebe, thank you for the great library.

I would like to use it on my MKRWifi 1010 board with Next-PM sensor , I tried to modify your PM serial Library and use it on my board but no luck. 
**If you can help me I will appreciate that.**
My platformio.ini looks like :  [https://docs.platformio.org/en/latest/boards/atmelsam/mkrwifi1010.html](url)
My example:
`
#include <PMserial.h>
SerialPM pms(NextPM, Serial1);  // PM Serial, RX, TX
void setup() {
  Serial.begin(9600);
 // Serial1.begin(9600, SERIAL_8N1, 13, 14);
  pms.init();                   // config serial port
}

void loop() {
  pms.read();                   // read the PM sensor
  Serial.print(F("PM1.0 "));Serial.print(pms.pm01);Serial.print(F(", "));
  Serial.print(F("PM2.5 "));Serial.print(pms.pm25);Serial.print(F(", "));
  Serial.print(F("PM10 ")) ;Serial.print(pms.pm10);Serial.println(F(" [ug/m3]"));
  delay(10000);                 // wait for 10 seconds
}`

The errors:
In file included from D:\Anass_PM\PMserial\PMserial.ino:2:0:

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_library1/PMserial.h:64:24: error: 'uart' has not been declared

   SerialPM(PMS sensor, uart &serial) : pms(sensor) {

                        ^~~~

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_library1/PMserial.h: In constructor 'SerialPM::SerialPM(PMS, int&)':

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_library1/PMserial.h:65:13: error: cannot convert 'int*' to 'Stream*' in assignment

     uart = &serial;

             ^~~~~~

D:\Anass_PM\PMserial\PMserial.ino: At global scope:

PMserial:7:29: error: cannot bind non-const lvalue reference of type 'int&' to an rvalue of type 'int'

 SerialPM pms(NextPM, Serial1);  // PM Serial, RX, TX

                             ^

In file included from C:\Users\Asus\OneDrive\Documents\ArduinoData\packages\arduino\hardware\samd\1.8.6\variants\mkrwifi1010/variant.h:165:0,

                 from C:\Users\Asus\OneDrive\Documents\ArduinoData\packages\arduino\hardware\samd\1.8.6\cores\arduino/delay.h:23,

                 from C:\Users\Asus\OneDrive\Documents\ArduinoData\packages\arduino\hardware\samd\1.8.6\cores\arduino/Arduino.h:81,

                 from sketch\PMserial.ino.cpp:1:

C:\Users\Asus\OneDrive\Documents\ArduinoData\packages\arduino\hardware\samd\1.8.6\cores\arduino/Uart.h:45:5: note:   after user-defined conversion: virtual Uart::operator bool()

     operator bool() { return true; }

     ^~~~~~~~

In file included from D:\Anass_PM\PMserial\PMserial.ino:2:0:

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_library1/PMserial.h:64:3: note:   initializing argument 2 of 'SerialPM::SerialPM(PMS, int&)'

   SerialPM(PMS sensor, uart &serial) : pms(sensor) {

   ^~~~~~~~

exit status 1
cannot bind non-const lvalue reference of type 'int&' to an rvalue of type 'int'

Thanks in advance.

avaldebe, 2020-04-15:
> I would like to use it on my MKRWifi 1010 board with Next-PM sensor , I tried to modify your PM serial Library and use it on my board but no luck.
> **If you can help me I will appreciate that.**
> My platformio.ini looks like : https://docs.platformio.org/en/latest/boards/atmelsam/mkrwifi1010.html

The `atmelsam` platform is currently not supported.
I'll need to see how much work it would be. Hopefully not much.

> #include <PMserial.h>
> SerialPM pms(NextPM, Serial1);  // PM Serial, RX, TX

Can you give me more information about this NextPM sensor.
It is not on the list of supported sensors.
If is not one of the Plantower PM sensors, there is not much I can do.

anass-naouss, 2020-04-17:
Hi avaldebe, thank you for your reply, if the problem is with the sensor, I can use one of the Plantower PM sensor. My big problem is with Arduino MKR.

avaldebe, 2020-04-17:
OK, I'll see what is needed for `atmelsam`.
Alas, I do not have any board with an `atmelsam` MCU, so I'll need your feedback.

avaldebe, 2020-04-17:
@anass-naouss 

I had no real problem compiling for the `mkrwifi101`, with a `PMSx003` sensor on `Serial1`.
Please try the [HardwareSerial][] example.

[HardwareSerial]: https://github.com/avaldebe/PMserial/tree/atmelsam/examples/HardwareSerial#arduino-mkrwifi1010

Or you can change the sensor on your example sketch, and try again

```c++
#include <PMserial.h>
SerialPM pms(PMSx003, Serial1);  //  RX:13 and TX:14
void setup() {
  Serial.begin(9600);
  pms.init();                   // config serial port
}

void loop() {
  pms.read();                   // read the PM sensor
  Serial.print(F("PM1.0 "));Serial.print(pms.pm01);Serial.print(F(", "));
  Serial.print(F("PM2.5 "));Serial.print(pms.pm25);Serial.print(F(", "));
  Serial.print(F("PM10 ")) ;Serial.print(pms.pm10);Serial.println(F(" [ug/m3]"));
  delay(10000);                 // wait for 10 seconds
}
```

anass-naouss, 2020-04-17:
Thank you for your time, with my exempl now I have no error but no value in serial Monitor
[image] 

With the HardwareSerial example, I have this error:
```
Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino MKR WiFi 1010"

HardwareSerial_New:12:29: error: no matching function for call to 'SerialPM::SerialPM(PMS, Serial_&)'

 SerialPM pms(PMSx003, Serial); // PMSx003, UART

                             ^

In file included from D:\Anass_PM\HardwareSerial_New\HardwareSerial_New.ino:3:0:

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:65:3: note: candidate: SerialPM::SerialPM(PMS, HardwareSerial&)

   SerialPM(PMS sensor, HardwareSerial &serial) : pms(sensor)

   ^~~~~~~~

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:65:3: note:   no known conversion for argument 2 from 'Serial_' to 'HardwareSerial&'

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:46:7: note: candidate: constexpr SerialPM::SerialPM(const SerialPM&)

 class SerialPM

       ^~~~~~~~

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:46:7: note:   candidate expects 1 argument, 2 provided

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:46:7: note: candidate: constexpr SerialPM::SerialPM(SerialPM&&)

C:\Users\Asus\OneDrive\Documents\Arduino\libraries\PMserial_uart/PMserial.h:46:7: note:   candidate expects 1 argument, 2 provided

D:\Anass_PM\HardwareSerial_New\HardwareSerial_New.ino: In function 'void loop()':

HardwareSerial_New:76:13: error: 'class SerialPM' has no member named 'has_temperature_humidity'

     if (pms.has_temperature_humidity() || pms.has_formaldehyde())

             ^~~~~~~~~~~~~~~~~~~~~~~~

HardwareSerial_New:76:47: error: 'class SerialPM' has no member named 'has_formaldehyde'

     if (pms.has_temperature_humidity() || pms.has_formaldehyde())

                                               ^~~~~~~~~~~~~~~~

HardwareSerial_New:78:24: error: 'class SerialPM' has no member named 'temp'

       Serial.print(pms.temp, 1);

                        ^~~~

HardwareSerial_New:81:24: error: 'class SerialPM' has no member named 'rhum'

       Serial.print(pms.rhum, 1);

                        ^~~~

HardwareSerial_New:84:24: error: 'class SerialPM' has no member named 'hcho'

       Serial.print(pms.hcho, 2);

                        ^~~~
exit status 1
no matching function for call to 'SerialPM::SerialPM(PMS, Serial_&)'
```

avaldebe, 2020-04-17:
@anass-naouss 

Sorry, I forgot to mention that you need `build_flags = -D USE_HWSERIAL1` on your configuration for the HardwareSerial example

https://github.com/avaldebe/PMserial/blob/4e2385f90105c5cdd484ad99f4983fb73d620af3/examples/HardwareSerial/platformio.ini#L52-L55

avaldebe, 2020-04-24:
@anass-naouss 

Does it work?

anass-naouss, 2020-04-24:
Hi avaldeb, thanks for your work.
I don't have one of Plantower PM sensor. I've tried to support it for Next-PM sensor but no luck. 
[image]
[Archive(1).zip](https://github.com/avaldebe/PMserial/files/4527363/Archive.1.zip)

 I found some information about Next-PM (attachment). Could you help me, should I open a new issue?

Best regards

avaldebe, 2020-04-24:
Thanks for the sensor datasheet. The messages from this sensor is very different to the Plantower sensors supported on this library. Sorry, you need to write your own code for reading this sensor.

The `NextPM` sensor uses UART 115200 8N1 (3.3V TTL).
The commands are described on Section 2.1.2. of the datasheet and the response message is described on Section 2.1.3. Based on that information, you can try the following sketch

```c++
// See datasheet Section 2.1.2. for more commands
const uint8_t
    msgLen = 3,
    mean10sec[msgLen] = {0x81, 0x11, 0x6E}, // 10 sec mean, 1 sec update 
    mean1min[msgLen]  = {0x81, 0x12, 0x6D}, // 1 min mean, 10 sec update 
    mean5min[msgLen]  = {0x81, 0x12, 0x6D}, // 5 min mean, 1 min update 
    temp_rhum[msgLen] = {0x81, 0x14, 0x6B}; // temperature and humidity

// See datasheet section 2.1.3 for message description
const uint8_t bufferLen = 16;
uint8_t buffer[bufferLen];
inline uint16_t buff2word(uint8_t n) { return (buffer[n] << 8) | buffer[n + 1]; }

void setup() {
  Serial.begin(9600);
  Serial1.begin(115200); // NextPM on Serial1  (RX=13; TX=14)
}

void loop() {
  // read the NextPM sensor
  Serial1.write(mean1min, msgLen); // request PM data
  Serial1.readBytes(&buffer, 16); // read PM data

  // decode PM message, see datasheet section 2.1.3
  float pm01 = buff2word(9)/10.0;
  float pm25 = buff2word(11)/10.0;
  float pm10 = buff2word(13)/10.0;

  Serial1.write(temp_rhum, msgLen); // request T/RH data
  Serial1.readBytes(&buffer, 8); // read T/RH data

  // decode T/RH message, see datasheet section 2.1.3
  float temp = buff2word(3)/100.0;
  float rhum = buff2word(5)/100.0;
// If you want the real ambient T/RH, you need to deactivate the heating function,
// and apply the correction coefficients to the raw data
// float temp = buff2word(3)*0.9754 - 4.2488;
// float rhum = buff2word(5)*1.1768 - 4.727;

  // print the results
  Serial.print(F("PM1.0 "));Serial.print(pm01, 1);Serial.print(F(", "));
  Serial.print(F("PM2.5 "));Serial.print(pm25, 1);Serial.print(F(", "));
  Serial.print(F("PM10 ")) ;Serial.print(pm10, 1);Serial.print(F(" [ug/m3], "));
  Serial.print(temp, 2);Serial.print(F("C, "));
  Serial.print(rhum, 2);Serial.println(F("%rh"));

  // wait for 10 seconds
  delay(10000);
}
```

If you want more help, I suggest you to start your own public repo with your sketch,
and add a link to the repo on a comment here.

For now I'll merge #12, which will close this issue

PR #12 (2020-04-24): HardwareSerial example for Arduino MKRWifi 1010 (atmelsam)
close #8
commit 2de7dba5 (2020-04-24): Merge pull request #12 from avaldebe/atmelsam

HardwareSerial example for Arduino MKRWifi 1010 (atmelsam), close #8
closed 2020-04-24 by avaldebe