avaldebe/PMserial #10: Support PMS5003S/ST/T

avaldebe, 2020-04-16:
> Do you plan on supporting PMS5003T ?

_Originally posted by @gbalbinot in https://github.com/avaldebe/PMserial/issues/4#issuecomment-544882132_

gbalbinot, 2020-04-16:
Thank you for the update! We have been using your code in your ESP32 and it is very stable and reliable. Soon we will use PMS5003T.

avaldebe, 2020-04-16:
Glad you found it useful.

I'm currently just collecting all the information to add support these sensors.
So far so good, but I can not promise to when it will be done.

In the meantime, please note that the temperature on the `PMS5003T`/`PMS5003ST` is a signed integer. This means that using the `PMS5003` hack I suggested earlier (`temp=pms.n5p0/10.0;rhum=pms.n10p0/10.0`) will give the wrong value for negative temperature.

Something like `temp=((int8_t)pms.n5p0)/10.0;` should give the right results, but it would be better if you can wait for popper support to be added.

Also as I do not have any of this sensor variants, I really appreciate if you could let me know if the new code works for you.

avaldebe, 2020-04-16:
@gbalbinot 

It went faster than expected, but as I wrote before I do not have any of S/T/ST variants,
so I can not test this properly.

I'll close this issue after updating the examples, but I would really appreciate if you could let me know if it works for your `PMS5003T`.

closed 2020-04-16 by avaldebe

4444monkey, 2023-06-09:
> > Do you plan on supporting PMS5003T ?
> 
> _Originally posted by @gbalbinot in [#4 (comment)](https://github.com/avaldebe/PMserial/issues/4#issuecomment-544882132)_
> 
> ## Message structure
> `message`	`PMS5003`	`PMS5003T`	`PMS5003S`	`PMS5003ST`
> 32 bits	32 bits	32 bits	40 bits
> header	4 bits	4 bits	4 bits	4 bits
> `42 4d 00 1c`	`42 4d 00 1c`	`42 4d 00 1c`	`42 4d 00 24`
> body	26 bits	26 bits	26 bits	34 bits
> 12 values, 1 reserved	13 values	13 values	15 values, 2 reserved
> checksum	2 bits	2 bits	2 bits	2 bits
> ## `PMSx003` sensor
> The `PMSx003` sensor type will infer the sensor type from the message header. As the `PMS5003T`/`PMS5003S` variants have the same header than the `PMS5003` they need to be declared explicitly on the `SerialPM` constructor.

I'm trying to facilitate the PMS sensor project recently and so glad to see your library.
Would you give me a guide how to modify the SerialPM struct to incorporate the PMS500T module that can collect the temp&hum? I barely understand it as a beginner for this part.
Appreciate that If you could proivde any suggestion.

4444monkey, 2023-06-09:
> Glad you found it useful.
>
> I'm currently just collecting all the information to add support these sensors. So far so good, but I can not promise to when it will be done.
>
> In the meantime, please [...]

I'm trying to facilitate the PMS sensor project recently and so glad to see your library.
Would you give me a guide how to modify the SerialPM struct to incorporate the PMS500T module that can collect the temp&hum? I barely understand it as a beginner for this part.
Appreciate that If you could proivde any suggestion.

avaldebe, 2023-06-12:
you can try something like this
``` c++
#include <PMserial.h>
SerialPM pms(PMS5003T, Serial);  // PMS5003T, on hardware serial

void setup() {
  Serial.begin(9600);
  pms.init();                   // config serial port
}

void loop() {
  pms.read();                   // read the PM sensor
  Serial.print(F("PM1.0 "));Serial.print(pms.pm01);Serial.print(F(", "));
  Serial.print(F("PM2.5 "));Serial.print(pms.pm25);Serial.print(F(", "));
  Serial.print(F("PM10 ")) ;Serial.print(pms.pm10);Serial.print(F(" [ug/m3]"));;Serial.print(F(", ")); 
  Serial.print(pms.temp, 1);Serial.print(F(" °C"));Serial.print(F(", "));
  Serial.print(pms.rhum, 1);Serial.println(F(" %rh"));
  delay(10000);                 // wait for 10 seconds
}
```

commit 3453d14f (2020-04-16): T/RH/HCHO output on examples, close #10