avaldebe/PMserial #11: library.properties is missing for Arduino IDE and CLI

jandrassy, 2020-04-17:
see https://arduino.github.io/arduino-cli/library-specification/#15-library-format-rev-22

avaldebe, 2020-04-17:
Thanks for the link.

avaldebe, 2020-04-17:
@jandrassy 

Can you please test that the library.properties is correct?

jandrassy, 2020-04-17:
you don't have Arduino IDE installed? :-)

the file is ok

and you should really test the library with some SAMD board. it compiles, but I don't have any sensor of this type.

closed 2020-04-17 by jandrassy

avaldebe, 2020-04-20:
> you don't have Arduino IDE installed? :-)

Nop, I use platformio/vscode. That way I get to use the same development environment for work (fortran, bash, python) and hobby (python, c++).

One added benefit of platformio for developing a library is that it can be easily integrated with a ci/cd service, like travis. I just did that, in the hope that it will help me to catch changes on the underlying Arduino core(s) and libraries such as with #6. Now every time I push changes all the examples get compiled with the latest version of the Arduino core(s) and libraries.