avaldebe/PMserial #13: Potential Error with has_number_concentration() and PMS5003ST

fmatray, 2020-11-19:
Hello,
I think there is an error with has_number_concentration() with PMS5003ST.
According to the datasheet, this sensor have concentration datas

I'm not sure yet, as I only have the 7003 for the moment.
I'll check when I receive the 5003ST, in 2 weeks.

Thanks.

avaldebe, 2020-11-19:
You are correct, The `PMS5003ST` has number concentrations, but `has_number_concentration` returns false

https://github.com/avaldebe/PMserial/blob/ebd4326b0cf27ebade2b87a917c5c826a8309ec5/src/PMserial.h#L124

This error also affects the `PMS5003S` and `PMS5003T` models.

Only the `PMS3003` lacks number concentrations, so the following patch should fix the problem

```diff
- inline bool has_number_concentration() { return (status == OK) && (pms == PLANTOWER_32B); }
+ inline bool has_number_concentration() { return (status == OK) && (pms != PMS3003); } 
```

avaldebe, 2020-11-19:
Hi @fmatray 

Just committed the patch I mentioned on my previous message.

I do not have a `PMS5003ST` to test this so I would appreciate if you could confirm that it works.
Please let me know if you find any other problem.
Once I have your conformation, I'll create a new tag/release.

Cheers,
Á.

fmatray, 2020-11-19:
Thanks a lot. 
I won't be able to check this correction before mid december. 

Cheers,
Frédéric

avaldebe, 2020-11-19:
No problem, the new version release can wait.

fmatray, 2020-11-25:
I received the 5003ST sooner than expected. 
And this point works.

avaldebe, 2020-11-25:
> I received the 5003ST sooner than expected.
> And this point works.

Great. Thanks for checking

commit bc9fa7f6 (2020-11-19): has_number_concentration on PMS5003ST,  fix #13
closed 2020-11-25 by avaldebe