avaldebe/PMserial #14: Sleep and wake mode

fmatray, 2020-11-19:
Hello,
I need to turn the sensor into sleep mode for a long time (several hours).
This is to prevent the fan to turn while the arduino is in deep sleep mode.
It's possible to low the set pin but I'm very limited with the amount pins. This method can save me a pin.

Is it possible to add this to your library please ?
Tested with a PMS7003.

const uint8_t
    msgLen = 7,
    //act[msgLen] = {0x42,0x4D,0xE1,0x00,0x01,0x01,0x71}, // set active mode
    slp[msgLen] = {0x42, 0x4D, 0xE4, 0x00, 0x00, 0x01, 0x73},  // sleep          **<-- UNCOMMENT**
    wak[msgLen] = {0x42, 0x4D, 0xE4, 0x00, 0x01, 0x01, 0x74},  // wake          **<-- UNCOMMENT**
    cfg[msgLen] = {0x42, 0x4D, 0xE1, 0x00, 0x00, 0x01, 0x70},  // set passive mode
    trg[msgLen] = {0x42, 0x4D, 0xE2, 0x00, 0x00, 0x01, 0x71};  // passive mode read

[...]

void SerialPM::sleep() {
  uart->write(slp, msgLen);  // sleep mode 
  uart->flush();
  delay(max_wait_ms * 2);
}
void SerialPM::wake() {
  uart->write(wak, msgLen);  // wake mode 
  uart->flush();
  delay(max_wait_ms * 2);
}

I found the issue #2 about this point, but it seems the code never been implemented.

Cheers, 
Frédéric

avaldebe, 2020-11-19:
> Is it possible to add this to your library please ?
> Tested with a PMS7003.

I would be happy to accept your PR.

avaldebe, 2020-11-19:
thanks for the PR

closed 2020-11-19 by avaldebe