avaldebe/PMserial #18: PM standard vs under atmospheric environment

szerwi, 2021-02-24:
Hi,

In PMS5003 datasheet I've found that the sensor is returning 2 different PM values. There is a value described as ```CF=1，standard particle``` and a value described as ```under atmospheric environment```. Do you know what is the difference between, for example PM10 standard and PM10 under atmospheric environment? Which value is being read by this library? Is there a possibility to read the other value using this library?

szerwi, 2021-02-24:
I have found out that in the newest version of the library it is possible to get both parameters from the sensor by defining tsi_mode to true or false in ```read(bool tsi_mode, bool truncated_num)``` function.
https://github.com/avaldebe/PMserial/blob/master/src/PMserial.cpp#L201-L206

So the remaining question is: What are the differences between standard and ATM and which one is better to use in monitoring air pollution in outside environment?

avaldebe, 2021-03-01:
As far as I understand the `CF=1` variables are "raw" estimates from the particle counts, assuming a fix particle radius and composition (likely used for used for factory calibration and quality control). The `ATM` variables (default) are compensated for "real" particles.

szerwi, 2021-03-01:
Thank you for the information. I have tested ATM values in comparison to CF=1 values and it seems that in low pollution conditions values are the same, but the higher pollution gets, the bigger the difference is. In high pollution environment, ATM values are lower than CF=1 values.

avaldebe, 2021-03-01:
I have observed 

> it seems that in low pollution conditions values are the same, but the higher pollution gets, the bigger the difference is. In high pollution environment, ATM values are lower than CF=1 values.

Yes, I have observed the same behavior.

It looks like I managed to answer your question, so I'll close this issue.
Please re-open it if my answer was lacking.

closed 2021-03-01 by avaldebe