avaldebe/PMserial #19: SoftwareSerial bug - local variable, dangling pointer

heX16, 2021-02-28:
I look at [this code](https://github.com/avaldebe/PMserial/blob/98ccdffcfd50219e522fd0830552d082bae824e7/src/PMserial.h#L88) and see an error.
The variable is local, its lifetime is limited to the area of the constructor. After exiting the constructor, the variable will be destroyed,  and the `uart` variable will contain [dangling pointer](https://en.wikipedia.org/wiki/Dangling_pointer). Copying its address should not prolong the lifetime.

I tried looking for some special case (among 19 methods of initialization, it's easy to forget about their special features) that affects the lifetime of a variable (1,2,3). But I didn't find anything. If my fears are in vain, please give me a link to a description of this case in the standard or explain how it works.

```
class SerialPM
{
protected:
  Stream *uart;  // hardware/software serial
....

  SerialPM(PMS sensor, uint8_t rx, uint8_t tx) : pms(sensor)
  {
    SoftwareSerial serial(rx, tx);
    uart = &serial;
    hwSerial = false;
  }
```

avaldebe, 2021-03-01:
I think you're, right. If I remember correctly, I tested this feature with actual HW when I implemented this feature, but I probably hit an undefined behavior of the uC tool-chain.

Will you be interested on submitting a patch/PR?

heX16, 2021-03-02:
> Will you be interested on submitting a patch/PR?

Yes! I have an idea to declare "Serial" outside and pass it to pointer initialization (it will be completely analogous to "HardwareSerial"). And make the incoming pointer type "[Stream](https://github.com/arduino/ArduinoCore-avr/blob/6ec80154cd2ca52bce443afbbed8a1ad44d049cb/libraries/SoftwareSerial/src/SoftwareSerial.h#L50)". This will allow you to use any "Serial" and even their non-standard implementations.
I'll add code and correct examples of working with "SoftwareSerial". Work with "HardwareSerial" will not change.

avaldebe, 2021-03-02:
> Yes! I have an idea to declare "Serial" outside and pass it to pointer initialization (it will be completely analogous to "HardwareSerial"). And make the incoming pointer type [...]

Awesome!

avaldebe, 2021-03-11:
fixed on #20

closed 2021-03-11 by avaldebe