avaldebe/PMserial #21: Once add "TridentTD_LineNotify" library to the project PMserial would give Exception(3)

cychrn, 2021-03-17:
I have tried using the example main.cpp in ProjectIO and the program and sensor works just fine.

BUT... after I have added TridentTD_LineNotify library into the project (just added the #include "TridentTD_LineNotify.h" line will already cause problem), it seems that when running, Exception(3) is thrown

the Exception is as follow

Exception (3):
epc1=0x40100e2d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40069bd8 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd40 end: 3fffffc0 offset: 0190
3ffffed0:  00000019 40262010 00000000 4020ec2d  
3ffffee0:  78534d50 20333030 53206e6f 00000040  
3ffffef0:  3fff1c5c 3fffea28 00000020 401010cf  
3fffff00:  00000a00 3fff1b3c 00000020 00000040  
3fffff10:  3fff1c5c 40205948 3fffea50 4021f709  
3fffff20:  00000003 40205a3c 00000020 4021f6f4  
3fffff30:  3ffe8cc2 00002580 00000041 40205802  
3fffff40:  00000003 3fffea28 3fffea00 40205980  
3fffff50:  4020e850 00000001 3fffea00 40205a56  
3fffff60:  4020e850 3ffefbe0 3ffe8cc0 3ffefd1c  
3fffff70:  3fffdad0 00000000 3ffefb1c 4020e17b  
3fffff80:  00000040 00000000 3ffefbe0 4020ecb0  
3fffff90:  3fffdad0 00000000 3ffefbe0 402054f1  
3fffffa0:  feefeffe feefeffe 3ffefcdc 4020ffa8  
3fffffb0:  feefeffe feefeffe 3ffe85e0 401013c5  
<<<stack<<<

Once i remove TridentTD_LineNotify library from the project, it just work once again.

I am using NodeMCU ESP8266 so I have made sure that

lib_deps = EspSoftwareSerial@>=6.7.1

is included into the platformio.ini as well

I have been trying everything out of my small knowledge but nothing seems to work... Could you please help to identify where it crash? I would really need to use this library along with TridentTD_LineNotify to notify PM2.5 to LINE Notify service.

Thank you very much!

avaldebe, 2021-03-18:
I can not look into it at the moment, please be patient.

> I am using NodeMCU ESP8266 so I have made sure that
> 
> lib_deps = EspSoftwareSerial@>=6.7.1
> 
> is included into the platformio.ini as well

`EspSoftwareSerial` is no longer needed, so it was removed on PR #20.
See plerup/espsoftwareserial#200 for details.

cychrn, 2021-03-18:
Oh i will remove it and try again and let you know soon. Thx for the reply

cychrn, 2021-03-19:
I have tried removing EspSoftwareSerial and the error still occur...

Please help check it out when you are available.

Thank you very much in advanced

cychrn, 2021-03-19:
May I know what does

Exception (3):
epc1=0x40100e2d epc2=0x00000000 epc3=0x00000000 excvaddr=0x40069bd8 depc=0x00000000

means? i'll try to trace the code of the included library and see if there's anything wrong with it.

Using these libraries alone has no problem at all...

cychrn, 2021-03-19:
After checking bit by bit, i found that the error occurs at this statement...

    static_cast<SoftwareSerial *>(uart)->begin(9600);

could there be anything crash from this statement? please advise

avaldebe, 2021-03-19:
> After checking bit by bit, i found that the error occurs at this statement...
> 
> ```
> static_cast<SoftwareSerial *>(uart)->begin(9600);
> ```
> 
> could there be anything crash from this statement? please advise

Looks like this is related to #19, which was fixed on #20.
I'll release a new version, so it is easier for you to update.

avaldebe, 2021-03-19:
I released version 1.1.1, it can take up to a day for the new version to show up on platfromio.
Please let me know if your problem persists.

cychrn, 2021-03-19:
I have tried 1.1.1 and it works out great now for all the libraries I'm using in the project!! Thank you very much!!!

avaldebe, 2021-03-19:
> I have tried 1.1.1 and it works out great now for all the libraries I'm using in the project!! Thank you very much!!!

Great news!

closed 2021-03-19 by avaldebe