avaldebe/PMserial #22: Need help getting Arduino Mega 2560 w/ PMS5003 to work

rtejada15, 2021-06-30:
FYI, I am a beginner with Arduino and coding in C++, I started working with Arduino stuff just 10 months ago. Since then I've been using the Arduino UNO that used a PMS5003, SD Card Breakout Board, and BME688, all of which using codes from the product page on Adafruit's website.

I ran into memory limitations with the Arduino UNO, thus I want to get the Plantower PMS5003 to work with the Arduino Mega and found this coding library. The Arduino Sketch I used was the HardwareSerial example found in this library and I did not edit it at all other than adding comments.

The wiring was as follows from sensor to MCU:
-VCC to 3.3V
-GND to GND
-TXD to TX0 (pin 1), so this should be going through Serial

When I run the code, this is the output in the serial monitor:

PMSx003 on HardwareSerial
BM⸮ pBM⸮ q Sensor read timeout

Do I need to make changes to the code and/or is the wiring wrong?

avaldebe, 2021-07-01:
I do not have access yo your code, so I can not tell you if this is a software problem.
However there are a few hardware issues you need to address in order to get the sensor to work realiably.

The [example's readme] has a section about the Arduino Mega 2560, wich says

```markdown
## [Arduino Mega 2560](https://docs.platformio.org/en/latest/boards/atmelavr/megaatmega2560.html)

**Note** This is a 5V board, and the PMS5003 RX/TX pins are 3.3V.
You need to add a logic level converter, or use a 3.3V version of this board.
```

[example's readme]: https://github.com/avaldebe/PMserial/tree/master/examples/HardwareSerial#arduino-mega-2560

The `PMS5003` needs 5V to operate the fan (and maybe the laser inside, I do not really know), but the RX/TX pins are only 3.3V tolerant. The `Mega 2560` is a 5V device, so you might damage the PM sensor if you connect it directly to the to the Mega.
You need a voltage level converter/shifter between the Mega and the PM sensor.

Also, this library operates the sensor on passive mode. This is to say that, it will ask the senor for one measurement at the time instead of having the sensor steam new measurements all the time. For this to work you need to wire both UART lines (RX and TX).
With this things in mind, here is how you connect the sensor.

| Mega 2560 |               | PMS5003 |
| -------- | :-----------: | ------- |
| 5V       |               | VCC     |
| GND      |               | GND     |
| TxN      | level shifter | RxD     |
| RxN      | level shifter | TxD     |

**Note** that the Mega Tx pin ends on the PM sensor's Rx pin and vice versa.

I hope this can help you, and please let me know if you manage to solve the problem.

avaldebe, 2021-08-23:
Hi @rtejada15 

It seems I managed to answer your question, so I'll close this issue.
If not the case, please do not hesitate to reopen this issue or open a new one.

closed 2021-08-23 by avaldebe