avaldebe/PMserial #23: Help me with Hardware Serial

carminelau, 2021-11-02:
Hi I have an esp32 and I connected a PMS A003 to the serial hardware (IO10, IO9) but it doesn't work, I always get "Sensor read timeout".
I used the HardwareSerial.ino sample code

avaldebe, 2021-11-03:
Hi @carminelau 

On some boards, like the [ESP32 MiniKit](https://github.com/avaldebe/PMserial/tree/master/examples/HardwareSerial#esp32-minikit), those pins are conected to the flash and using them might crash your board.

Other option is that your RX/TX lines are swapped. You need to connect them like this:

- ESP32 TX --> RX PMSA003
- ESP32 RX <-- TX PMSA003

Hope this helps

carminelau, 2021-11-03:
My board is ESP32 WROVER-B can you help me ?

carminelau, 2021-11-03:
[image]

This is my esp32, i used the GPIO 9 and GPIO10 but my esp crash always

avaldebe, 2021-11-03:
> This is my esp32, i used the GPIO 9 and GPIO10 but my esp crash always

Your problem might be related to this issue:

> UART1 defaults to pins 9 and 10 which are used by the onboard flash. Either use UART2 on pins 16 and 17 or define the pins you want to use when starting UART1

_Originally posted by @me-no-dev in https://github.com/espressif/arduino-esp32/issues/148#issuecomment-275635651_

Is this crash related from the use of this library?
Or it always crash when using `Serial1`?
Can you try using `Serial2` as in the example?

carminelau, 2021-11-03:
Always crash when using Serial1, Serial2 or istanziate with RX and Tx

carminelau, 2021-11-03:
I use this code for testing...

#include <PMserial.h>
SerialPM pms(PMSA003, Serial1);  // PMSx003, UART   RX GPIO9 TX GPIO10

void setup() {
  Serial.begin(9600);
  pms.init();                   // config serial port
}

void loop() {
  pms.read();                   // read the PM sensor
  Serial.print(F("PM1.0 "));Serial.print(pms.pm01);Serial.print(F(", "));
  Serial.print(F("PM2.5 "));Serial.print(pms.pm25);Serial.print(F(", "));
  Serial.print(F("PM10 ")) ;Serial.print(pms.pm10);Serial.println(F(" [ug/m3]"));
  delay(10000);                 // wait for 10 seconds
}

can you help me ? i'm really tired.

avaldebe, 2021-11-03:
Just to be sure, you have tied with `Serial2`, l like in the following?

``` c++
#include <PMserial.h>
SerialPM pms(PMSA003, Serial2);  // PMSx003, UART   RX GPI16 TX GPIO17
...
```

carminelau, 2021-11-03:
No, I honestly don't know how to enable Serial2.

avaldebe, 2021-11-03:
just change the line as I showed on my previous message and connect the sensor the GPIO 16 and 17

carminelau, 2021-11-03:
if I want to use the pins I said GPIO9 and GPIO10 how should I do?

avaldebe, 2021-11-04:
> if I want to use the pins I said GPIO9 and GPIO10 how should I do?

You first need to check that the pins are usable.
Maybe you can first try with a simple echo sketch, like this one:

``` c++
void setup() {
  Serial.begin(9600);
  Serial1.begin(9600);

  Serial.println("hello from Serial");
}

void loop() {
  if Serial.available() {
    Serial1.write(Serial.read());
  } 
  if Serial1.available() {
    Serial.write(Serial1.read());
  } 
}
```

and connect GPIO9 and GPIO10 with a jumper wire. You should get the message `"hello from Serial"` on the serial monitor (or whatever the Arduino IDE calls it).

heX16, 2021-11-06:
> I use this code for testing...
If you linking Serial1 directly, you must init then.

```
Serial1.begin(9600, SERIAL_8N1, _RX_, _TX_);
```
See more: https://github.com/avaldebe/PMserial/tree/master/examples/SoftwareSerial#esp32-minikit

avaldebe, 2021-11-10:
@carminelau 

did you get the sensor working on Serial1?

carminelau, 2021-11-10:
No, the serial1 it is not usable in my esp32

avaldebe, 2021-11-10:
> No, the serial1 it is not usable in my esp32

Did you try to use different pins, as @me-no-dev  and @heX16 suggested?