avaldebe/PMserial #30: Problem with init

carminelau, 2022-09-26:
Hi I made the init and after a few seconds about 10 I'm going to make the first reading. The problem is that sometimes the reading of the three Particulate values ​​is 0.

avaldebe, 2022-09-28:
What sensor model are you using?
Maybe you could check your senor wiring using a USB serial adaptor and my Python library [PyPMS]?

[PyPMS]: https://avaldebe.github.io/PyPMS/

avaldebe, 2023-09-18:
Hi @carminelau 

It is close to a year since you open this issue, and you have not answered my follow up questions. Without that info I can not help you to debug your problem:

I hope you managed to find the problem since, so I'll close the issue.
Please re-open it if this was not the case.

closed 2023-09-18 by avaldebe