avaldebe/PMserial #32: Doesn`t work on esp32 wroom 32u

aminxkamali, 2024-01-19:
Hello my friend
I have a project that i use pms5003 
Also i use gps,mq131,mq7 and LED 64*128 
I CONNECT VCC of pms5003 wire to 5v esp3 and connect gnd to gnd
Then connect rx and tx of pms to tx and rx of esp32
I try any code that you know but i have a bad problem (very very bad)
When i run my code and run the project on my esp
In buad monitor i get something bad
Some answers that i dont undrestand(i just use 9600 baud but my pms respons on 115200 baud)
I send a pic that you can see my problem.
I try all ways for solve this problem but i get bad response on monitor bauds that is not 9600.
[image]
Can you help me?

aminxkamali, 2024-01-19:
When i delete the library of pms and the coding of library 
Everythings back to normal !!!!

avaldebe, 2024-01-19:
can you show me a minimal version of your sketch/program?